mariusgreuel/avrdude #1: Wonderful approach, is a Linux version available?

ArminJo, 2021-01-05:
Thank you for including the micronucleus bootloader int avrdude. 🥇 

I (and others) currently have the problem with the old linux binary of the uploader not accepting new firmware versions.
```
Device reports version as: 2.2
Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
```
https://github.com/micronucleus/micronucleus/issues/195#issuecomment-753538691

So it would be nice to have the avrdude integation also available for linux 😀 
Thanks
Armin

BTW. you may want to activate GitHub Discussions
https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/enabling-or-disabling-github-discussions-for-a-repository#enabling-or-disabling-discussions-for-your-repository

mariusgreuel, 2021-01-12:
Armin: The micronucleus bootloader is the only feature I actually backported to Linux. Can you please try the build instructions here (its for Ubuntu) at provide feedback?
https://github.com/mariusgreuel/avrdude#building-avrdude-for-linux

Thanks,
Marius

ArminJo, 2021-02-04:
Hi Marius,
before I try to port it to Linux, could there be a parameter for -x wait like -x wait=60 and an output:
avrdude.exe: No device found, waiting for device **to be plugged in (will time out in 60 seconds)** ...

And what are the chances to integrate your valuable changes in the upstream version, removing the need of the micronucleus binary for Arduino?

mariusgreuel, 2021-02-07:
>  could there be a parameter for -x wait like -x wait=60
The time-out is a nice addition. I just added it.

> what are the chances to integrate your valuable changes in the upstream version
I was planning on kicking back some changes, however, I was hoping for some sort of feedback first. Some changes are Windows only, others I expect to be controversial. For starters, I created an upstream branch, where I picked out the changes for micronucleus.

As you seem to be on Linux, would you be so kind to try the upstream branch changes? I updated the instructions here
https://github.com/mariusgreuel/avrdude#building-avrdude-for-linux
you just need to replace the windows branch with upstream.

closed 2022-10-12 by ArminJo