mariusgreuel/avrdude #7: Windows Based GPIO support

rupeshshet, 2022-01-13:
HI I am looking to use avrdude.exe in docker image for IotEdge written in c# and running on linux. Can this support GPIO model in future?

mariusgreuel, 2022-01-13:
The Windows version does not support GPIO. However, there is Linux GPIO support, if that helps:
 https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Linux

mcuee, 2022-05-21:
> Windows Based GPIO support

Without a generic GPIO driver under Windows, it is basiclaly NO Go for this idea. 

You will have better luck with other programmers like FTDI based programmer.

Last time avrdude got the old giveio driver which does not work for newer version of Windows. It was removed in this commit o 22 Dec 2021.
https://github.com/avrdudes/avrdude/commit/60cb5480756025b7354b01fe554cf959a1966bbc