mariusgreuel/avrdude #11: Writing fusebits to the MCU from a file does not working

masya-chel, 2022-07-22:
I am trying to write fusebits to the MCU from a file, this function does not working - an error is displayed. See the screenshots below.

[image]

[image]

`avrdude.exe: Device signature = 0x1e9307 (probably m8)`
`avrdude.exe: reading input file "C:\CLI\AVRdude_CLI\fw_write\high.txt"`
`avrdude.exe: fileio: invalid operation=0`
`avrdude.exe: read from file 'C:\CLI\AVRdude_CLI\fw_write\high.txt' failed`

If I try to read fusebits from the MCU to a file, then all works correctly - files with the names "high" and "low" are created in the folder, and the value of fusebits is contained inside the files. See the screenshots below.

[image]

[image]

If I try to write fusebits to the MCU directly from the console, then fusebits are written correctly. See the screenshot below.

[image]

Tested on avrdude versions: 6.3; 6.4; 7.0. The function of writing fusebits to the MCU from a file does not work.

mariusgreuel, 2022-07-27:
As the issue does not seem to be Windows specific, can you please post the issue here:
https://github.com/avrdudes/avrdude/issues

mcuee, 2024-01-27:
I believe the issue has been fixed in avrdude 7.2 release.
https://github.com/avrdudes/avrdude/releases/tag/v7.2