mariusgreuel/avrdude #12: USBasp firmware incompatibility

Ed8562, 2022-09-17:
Doesn't recognize USBasp loaded with firmware version 1.09 from:
https://github.com/dioannidis/usbasp

D:\Avrdude>avrdude -q -q -C avrdude.conf -p atmega328p -c usbasp -e -U lock:w:0x3F:m -U efuse:w:0xFD:m -U hfuse:w:0xDF:m -U lfuse:w:0xFF:m
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'

(avrdude version 6.3.1.1-windows, URL: <https://github.com/mariusgreuel/avrdude>)

Savannah AVRdude for Windows (6.3-20190619 and later) work fine with the v1.09 firmware.

Windows driver:  libusbK

Ed8562, 2022-09-18:
I can't explain it but now firmware V1.09 is suddenly working with AVRdude version 6.3.1.1-windows.
The only difference I've noted is the way USBasp appears under Zadig.  When I was having issues it showed as two devices:
USBasp (Interface 0)
USBasp (Interface 1)

Now it simply shows as 'USBasp'.  I'm using a 4-port USB hub for my connections.  As I now can't duplicate the fault, may as well close the issue unless someone has something to add.

mcuee, 2024-01-27:
Reference: you can use latest avrdude 7.2 release but with mingw build.
* https://github.com/avrdudes/avrdude/issues/968

Binary download:
https://github.com/mcuee/avrdude/releases/tag/v7.2