mariusgreuel/avrdude #13: Very slow writing speed

impnet, 2022-10-28:
Hi, i just update from v6.3.1.1 to v7.0 but there is problem with writing speed.

raspB975, 2024-01-25:
Hi! I'm having the same issue with writing/reading LOW speed via FTDI. How are you solving the problem?

impnet, 2024-01-26:
> Hi! I'm having the same issue with writing/reading LOW speed via FTDI. How are you solving the problem?

Nope, did not find any workaround.  I switch back to version 6.3.1.1.

raspB975, 2024-01-26:
Solution:

This is a known version of the official MSVC build.
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-programmers

Use the MinGW version, but you need to switch the driver to WinUSB using Zadig.
https://github.com/mcuee/avrdude/releases/tag/v7.2 (avrdude-v7.2_mingw64.zip)
https://github.com/arduino/avrdude-packing/releases/tag/7.2-arduino.1
Now v7.2 working fast.  

> > Hi! I'm having the same issue with writing/reading LOW speed via FTDI. How are you solving the problem?
> 
> Nope, did not find any workaround. I switch back to version 6.3.1.1.

impnet, 2024-01-26:
> Solution:
>
> This is a known version of the official MSVC build. https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-programmers
>
> Use the MinGW version, but you need [...]

https://github.com/mcuee/avrdude/releases/tag/v7.2
Does not work. 
```
avrdude error: cannot open ftdi device: usb_open() failed
avrdude error: unable to open programmer arduino-ft232r on port usb
```
If there is need to instal special drivers via some another software, then it is useless.

https://github.com/arduino/avrdude-packing/releases/tag/7.2-arduino.1
Does not work. 
```
avrdude error: cannot open ftdi device: usb_open() failed
avrdude error: unable to open programmer arduino-ft232r on port usb
```

raspB975, 2024-01-26:
My config file
```
#------------------------------------------------------------
programmer
  id    				= "ftbb";
  desc  				= "FT232R Synchronous BitBang";
  prog_modes            = PM_TPI | PM_ISP;
  connection_type       = usb;
  type                  = "ftdi_syncbb"; #	"ftdi_syncbb";
  usbvid                = 0x0403;
  usbpid                = 0x6001;
  usbdev                = "A";  
  sdi  					= 4;  			# DTR(2 PIN FT232R) (9 PIN ON board )  MISO 
  sck   				= 2;  			# RTS(3 PIN FT232R) (7 PIN ON board )   
  sdo  					= 7;  			# RI (6 PIN FT232R) (1 PIN ON board )  MOSI   
  reset 				= 1;  			# RXD(5 PIN FT232R) (5 PIN ON board )  
  buff 					= ~0;  			# TXD(1 PIN FT232R) (GATE pins!)   
;
```

raspB975, 2024-01-26:
[image]

mcuee, 2024-01-27:
Reference: this is a known issue of avrdude MSVC build. Work-around is to use mingw build (you need to switch driver to WinUSB).
* #1253

Binary download:
https://github.com/mcuee/avrdude/releases/tag/v7.2