mariusgreuel/avrdude #14: FT4232H Channel B not working

PhySix66, 2022-12-15:
Programer:
FT4232H IC - module: FT-MOD-4232HUB
Target: ATMega1284P

Changing the line:
usbdev     = "A";
to
usbdev     = "B";

Has no effect. The FT4232H still communicates via PORTA (ChannelA). However in the LOG (verbosity lvl 4) it sates that PORTB (ChannelB) is in use:
"I avrftdi_open(714): Using device VID:PID 0403:6011 and SN '(null)' on interface B."
...
"D set_pin(243): Setting pin 4 (BDBUS4) as RESET: low (high active)"

mcuee, 2024-01-27:
avrdude main line should work.
https://github.com/avrdudes/avrdude/releases/tag/v7.2

PhySix66, 2024-02-02:
> avrdude main line should work. https://github.com/avrdudes/avrdude/releases/tag/v7.2

avrdude 7.2 solved(fixed) the problem above. Thank You mcuee!
Now I'm able to use PortB on the FT-MOD-4232HUB module.
And tested this on the mikro click mikroe-1421 module (FT2232H chip) as well with success!
Note: Only one downside with that module is that the RST pin is connected to the C-Channel (BCBUS0 / Pin0 on PortC) so I had to use the ChipSelect Pin(BDBUS3 / Pin3 on PortB).
Closing Issue with Problem Solved via avrdude 7.2! (Hopefully new versions also.)

closed 2024-02-02 by PhySix66